CadQuery/ocp-build-system #8: cadquery-ocp 7.6.3a0 not working on Windows 7

polycorn, 2023-01-16:
The cadquery-ocp 7.6.3a0 version included a compiled vtk 9.1, which used the tbb12-XXX.dll. This dll had a call to Kernel32.dll for function "GetCurrentThreadStackLimits", and windows7 does not have it. It is only available on Windows 8+.
The offical vtk 9.1 package does not have this depenedency, thus, it can be executed on windows 7.
The old cadquery-ocp 7.5.3.0's tbb dll does not have this function call.

fpq473, 2023-01-16:
The win_amd64 wheel was built by running ([link](https://github.com/CadQuery/ocp-build-system/blob/daed1f381700a20ac5af0e311e3c7442f4fb5238/.github/workflows/build-wheel.yml#L54)):
```
conda install -n cadquery-ocp -y ocp=7.6.* vtk=9.1.0
```
and then packaging the tbb12.dll that was installed.

I can see that `https://conda.anaconda.org/conda-forge/win-64/tbb-2021.6.0-h91493d7_0.tar.bz2` was installed and I can see "GetCurrentThreadStackLimits" in tbb12.dll.  So I'm guessing that it's not a vtk problem, it's just that in Oct 2022, when the wheel was built, there was a newer version of tbb, and that was packaged into the wheel.

I don't have access to a Windows machine so it's hard for me to propose a fix, but @polycorn, if you are a conda user, perhaps you can find an install command that installs ocp 7.6, vtk 9.1, and **doesn't install an incompatible version of tbb**.  Wheels are built on Windows Server 2022 or 2019 so ideally you'd check on that, but I imagine checking on Windows 10 might be good enough.

If you do this, please paste the output of the install command and your Windows version.

polycorn, 2023-01-19:
> 

Thanks for the reply ! I am not familiar with conda, but i will look into it later. 
From what I have seen, https://anaconda.org/conda-forge/tbb/files has a version https://anaconda.org/conda-forge/tbb/2021.5.0/download/win-64/tbb-2021.5.0-h91493d7_5.tar.bz2, this tbb library is available to win7.

jmwright, 2024-01-02:
We have moved well past this version of OCP and VTK, so I'm going to close this. Please reopen if this issue persists with the latest wheels.

closed 2024-01-02 by jmwright