CadQuery/ocp-build-system #16: Support python 3.12

ankurvdev, 2023-12-06:
The current packages on pypi only support 3.11
please update the packages to support 3.12

jmwright, 2023-12-06:
Thanks @ankurvdev . I'm hoping to add Python 3.12 support with the 7.7.2 release, but with Python 3.11 we had to wait for some of our dependencies to catch up. My guess is that we'll run into the same issue again.

jmwright, 2023-12-29:
@ankurvdev and everyone else:

Please manually test the wheels [here](https://github.com/CadQuery/ocp-build-system/actions/runs/7353258115), especially the Python 3.12 package. It worked fine for me on Ubuntu 22.04. However, the nlopt package seems to have a problem with Python 3.12, so it is probably not possible to get a full working CadQuery environment.

jmwright, 2024-01-02:
The OCP 7.7.2 wheels have been published, including for Python 3.12.

closed 2024-01-02 by jmwright