CadQuery/ocp-build-system #20: Linux aarch64 (ARM64) Wheels

jmwright, 2024-01-10:
We are out of project storage on PyPI for now and so I cannot upload these wheels, but I will post them here in case people want to test them.

https://www.dropbox.com/scl/fo/bp7df3u6ewe6rjijh0xwq/h?rlkey=hkbp4705uyviky5mdh1snfg7b&dl=0

It was built on a Pine64 RockPro64 running Debian 12. I tried building on Debian 11 to cover older systems too, but patchelf was too old on Debian 11. You will need to grab the correct file for your Python version. After installing the wheel locally I was able to run the following.
```python
import OCP
print(OCP.__version__)
```
If these work well for other people I will consider turning the RockPro64 into a hosted runner for GitHub.

thasti, 2024-02-10:
These wheels work fine for me (tested on Asahi Fedora Remix on a MacBook M2 Air). I would see a bit benefit in  having those available on PyPi to simplify installing CadQuery. Can the storage limitations on PyPi be resolved, e.g. is this a matter of financial support?

jmwright, 2024-02-10:
> Can the storage limitations on PyPi be resolved, e.g. is this a matter of financial support?

I don't think so. I requested the size increase as an addendum to another request, so it may have been missed. I'll file a full request from scratch to see what happens.

If we do eventually have to self-host our Python wheels, then there will be server cost involved. I doubt we are there yet though.

jmwright, 2024-02-12:
FYI: https://github.com/pypi/support/issues/3631

jmwright, 2024-02-19:
The Linux aarch64 wheels have been attached to the release [here](https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.0). It is possible to do a pip install of the wheels on that release by copying the URL for the file that matches your OS and Python version and then doing `pip install URL_OF_CORRECT_WHEEL_FILE`.

jmwright, 2024-08-09:
@jdegenstein I am having some trouble with the Linux ARM64 wheels, and I wonder if you might have some insight. The wheels build without errors using Anaconda locally, but they include the tag for Python 3.10 internally (in *dist-info/WHEEL), even though all the .so libs are named with the correct Python version. The native Python version on the host system is 3.11, so I am not sure where this is coming from. When I try to import any version of the wheel other than Python 3.10, I get an error.

```python
(ocp-3.9) $ pip install ./cadquery_vtk-9.2.6-cp39-cp39-linux_aarch64.whl 
Processing ./cadquery_vtk-9.2.6-cp39-cp39-linux_aarch64.whl
Installing collected packages: cadquery-vtk
Successfully installed cadquery-vtk-9.2.6
(ocp-3.9) $ python
Python 3.9.19 | packaged by conda-forge | (main, Mar 20 2024, 13:51:08) 
[GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import vtk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jwright/mambaforge/envs/ocp-3.9/lib/python3.9/site-packages/vtk.py", line 4, in <module>
    from vtkmodules.vtkCommonCore import *
ImportError: libvtkWrappingPythonCore3.9-9.2.so: cannot open shared object file: No such file or directory
```

When I open the wheels and check the `vtkmodules` directory, the lib* files are not being copied into the wheel for any other Python version except 3.10. Any thoughts on why this might be happening? The local VTK build script is [here](https://github.com/CadQuery/ocp-build-system/blob/main/local-vtk-build.sh).

jmwright, 2024-08-10:
@jdegenstein It looks like the local build script is grabbing setuptools from the wrong Python version for some reason. I am trying to fix that now.

jmwright, 2024-08-15:
Anyone who is interested in ARM64 Linux builds, please test the following. It is a stepping stone to publishing the next release of cadquery-ocp.

```
 pip install -i https://test.pypi.org/simple/ cadquery-vtk==9.2.6
```

Then try running the following.

```
$ python
>>> import vtk
```

jmwright, 2024-09-06:
Please download the correct wheel for your platform and Python version and do a test installation. Once these have been tested, I will publish them to PyPI. During the installation, cadquery-vtk will be downloaded from PyPI since it is no longer embedded inside the OCP wheel.

https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.1

jmwright, 2024-09-10:
Linux aarch64 wheels are available now on PyPI.

https://pypi.org/project/cadquery-ocp/

closed 2024-09-10 by jmwright

jmwright, 2024-09-11:
There was an issue with the VTK dependency that is causing this wheel not to work properly for certain use cases, so I am reopening this issue.

jmwright, 2025-12-18:
Linux ARM64 should be returning with the next release.