CadQuery/ocp-build-system #23: cadquery and vtk compatibility

msavtchouk-pf, 2024-03-03:
Hi all
We are using the latest pip-installable cadquery and we also recently added a new dependency (vtk==9.3.0)
Just importing those two in the same file:
 `import cadquery
import vkt 
`
Results in the following error:
`ImportError: Initialization failed for vtkWebCore, not compatible with vtkmodules.vtkCommonCore`

Since that the vtk version, used by cadquery is incompatible with vtk==9.3.0.
I could not find any pointers how to resolve this so far. Any help would be appreciated!

jmwright, 2024-03-21:
@msavtchouk-pf Can you manually install the appropriate wheel artifact from [here](https://github.com/CadQuery/ocp-build-system/actions/runs/8367440736) and see if it fixes your problem (and if it even works correctly for you)? I removed the embedded version of VTK that we were building. I tested the Python 3.10 wheel on Debian and that at least works.

mathstuf, 2024-04-27:
FWIW, this error in VTK comes when a VTK Python module is loaded without VTK's Python glue code being initialized. Where did this VTK come from?

jmwright, 2024-04-29:
@mathstuf It is built from source in [this workflow](https://github.com/CadQuery/ocp-build-system/blob/main/.github/workflows/build-vtk.yml).

mathstuf, 2024-05-09:
I think this is missing `auditwheel` (Linux) and `delvewheel` (Windows) at least to fix up the wheels to contain the appropriate VTK libraries. It seems like some other VTK build may be getting loaded via `PATH` or `LD_LIBRARY_PATH`.