CadQuery/ocp-build-system #24: version 7.7.2 on pypi not being found for linux

shimwell, 2024-03-13:
Hi all

I've been trying to install cadquery-ocp 7.7.2 with pip on a regular ubuntu machine and not being able to install version 7.7.2 even though I can see 7.7.2 in the releases on [pypi](https://pypi.org/project/cadquery-ocp/7.7.2/#files) and on [this repo releases](https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.0).

in a python 3.12 environment no versions are found
[image]

in a python 3.11 environment 7.7.0 and 7.7.0a versions are found
[image]

in a python 3.10 environment several older versions are found but not 7.7.2
[image]

I've also tried installing the wheels directly from the repo  releases and from pypi urls

pip install from repo release wheel url in python 3.12 env
[image]

pip install from pypi release wheel url in python 3.12 env
[image]

I've also tried with the ```--pre``` comand for pip but no luck.

Am I doing something wrong or is the 7.7.2 release somehow different to the 7.7.0 release?

Many thanks

jmwright, 2024-03-13:
@shimwell That laptop is not ARM based, is it? Also, make sure pip is updated:

```
pip install --upgrade pip
```

shimwell, 2024-03-14:
Sorry I should have mentioned I'm on the newest pip version 24.0 and it is a normal x86 cpu (not arm).

shimwell, 2024-03-14:
Ah it just worked in a docker image (Ubuntu:latest) so it must be something on my local setup

shimwell, 2024-03-14:
I've not updated my OS to Ubuntu 22.04 and everything is installing nicely. I shall close this issue and I think it is safe to assume user error or local environment issues

closed 2024-03-14 by shimwell