CadQuery/ocp-build-system #28: Failing tests on Windows with GH artifact wheels

jdegenstein, 2024-04-22:
Using an appropriate wheel from https://github.com/CadQuery/ocp-build-system/actions/runs/8367440736

I created a conda environment and discovered that 2 tests from CadQuery/cadquery are failing. Here is my setup process:
```sh
conda create -n env1 python=3.11
conda activate env1
pip install cadquery_ocp-7.7.2-cp311-cp311-win_amd64.whl # this also installs vtk=9.3.0 on my system
pip install cadquery
pip install ipython pytest
git clone https://github.com/CadQuery/cadquery.git
cd cadquery
python -m pytest
```
fails 2 tests and returns:
```py
___________________________________________________________________ TestCadQuery.test_toVtk ____________________________________________________________________

self = <tests.test_cadquery.TestCadQuery testMethod=test_toVtk>

    def test_toVtk(self):

        from vtkmodules.vtkCommonDataModel import vtkPolyData

        f = Face.makePlane(2, 2)
        vtk = f.toVtkPolyData(normals=False)

        assert isinstance(vtk, vtkPolyData)
>       assert vtk.GetNumberOfPolys() == 2
E       assert 0 == 2
E        +  where 0 = <built-in method GetNumberOfPolys of vtkmodules.vtkCommonDataModel.vtkPolyData object at 0x0000017172998B20>()
E        +    where <built-in method GetNumberOfPolys of vtkmodules.vtkCommonDataModel.vtkPolyData object at 0x0000017172998B20> = <vtkmodules.vtkCommonDataModel.vtkPolyData(0x0000017155FF6E80) at 0x0000017172998B20>.GetNumberOfPolys

tests\test_cadquery.py:5634: AssertionError
--------------------------------------------------------------------- Captured stderr call --------------------------------------------------------------------- 2024-04-22 15:12:23.713 (  18.739s) [B3557CCDB772DC2B]vtkDemandDrivenPipeline:677    ERR| vtkCompositeDataPipeline (000001717144C740): Input port 0 of algorithm vtkTriangleFilter (00000171717DE5C0) has 0 connections but is not optional.
____________________________________________________________________ test_assy_vtk_rotation ____________________________________________________________________

tmpdir = WindowsPath('C:/Users/johnd/AppData/Local/Temp/pytest-of-johnd/pytest-2/out1')

    def test_assy_vtk_rotation(tmpdir):

        v0 = Vertex.makeVertex(1, 0, 0)

        assy = Assembly()
        assy.add(
            v0, name="v0", loc=Location(Vector(0, 0, 0), Vector(1, 0, 0), 90),
        )

        fwrl = Path(tmpdir, "v0.wrl")
        assert not fwrl.exists()
        assy.save(str(fwrl), "VRML")
        assert fwrl.exists()

        matched_rot = False
        with open(fwrl) as f:
            pat_rot = re.compile("""rotation 1 0 0 1.5707963267""")
            for line in f:
                if m := re.search(pat_rot, line):
                    matched_rot = True
                    break

>       assert matched_rot
E       assert False

tests\test_exporters.py:835: AssertionError
--------------------------------------------------------------------- Captured stderr call --------------------------------------------------------------------- 2024-04-22 15:12:29.652 (  24.678s) [B3557CCDB772DC2B]vtkDemandDrivenPipeline:677    ERR| vtkCompositeDataPipeline (0000017157D868C0): Input port 0 of algorithm vtkTriangleFilter (000001718E5DCAB0) has 0 connections but is not optional.
```

jmwright, 2024-04-22:
@jdegenstein Is this on Windows 11?

jdegenstein, 2024-04-22:
This is on Windows 10

jmwright, 2024-04-24:
@jdegenstein I can reproduce on Debian 12 with the wheel with mangling disabled, and Linux gives some extra info.
```python
_______________________________________________________________________________ TestCadQuery.test_toVtk _______________________________________________________________________________

self = <tests.test_cadquery.TestCadQuery testMethod=test_toVtk>

    def test_toVtk(self):
    
        from vtkmodules.vtkCommonDataModel import vtkPolyData
    
        f = Face.makePlane(2, 2)
        vtk = f.toVtkPolyData(normals=False)
    
        assert isinstance(vtk, vtkPolyData)
>       assert vtk.GetNumberOfPolys() == 2
E       assert 0 == 2
E        +  where 0 = <built-in method GetNumberOfPolys of vtkmodules.vtkCommonDataModel.vtkPolyData object at 0x7f2f607256c0>()
E        +    where <built-in method GetNumberOfPolys of vtkmodules.vtkCommonDataModel.vtkPolyData object at 0x7f2f607256c0> = <vtkmodules.vtkCommonDataModel.vtkPolyData(0x562feadfc2c0) at 0x7f2f607256c0>.GetNumberOfPolys

tests/test_cadquery.py:5634: AssertionError
-------------------------------------------------------------------------------- Captured stderr call ---------------------------------------------------------------------------------
2024-04-24 11:15:27.995 (  13.048s) [    7F2FB9744740]vtkDemandDrivenPipeline:675    ERR| vtkCompositeDataPipeline (0x562fea862810): Input port 0 of algorithm vtkTriangleFilter (0x562fea6424d0) has 0 connections but is not optional.
_______________________________________________________________________________ test_assy_vtk_rotation ________________________________________________________________________________

tmpdir = PosixPath('/tmp/pytest-of-jwright/pytest-5/out1')

    def test_assy_vtk_rotation(tmpdir):
    
        v0 = Vertex.makeVertex(1, 0, 0)
    
        assy = Assembly()
        assy.add(
            v0, name="v0", loc=Location(Vector(0, 0, 0), Vector(1, 0, 0), 90),
        )
    
        fwrl = Path(tmpdir, "v0.wrl")
        assert not fwrl.exists()
        assy.save(str(fwrl), "VRML")
        assert fwrl.exists()
    
        matched_rot = False
        with open(fwrl) as f:
            pat_rot = re.compile("""rotation 1 0 0 1.5707963267""")
            for line in f:
                if m := re.search(pat_rot, line):
                    matched_rot = True
                    break
    
>       assert matched_rot
E       assert False

tests/test_exporters.py:835: AssertionError
-------------------------------------------------------------------------------- Captured stderr call ---------------------------------------------------------------------------------
2024-04-24 11:15:32.150 (  17.203s) [    7F2FB9744740]vtkDemandDrivenPipeline:675    ERR| vtkCompositeDataPipeline (0x562feb3ca760): Input port 0 of algorithm vtkTriangleFilter (0x562feb3b51b0) has 0 connections but is not optional.
```

jmwright, 2024-04-24:
@jdegenstein can you try VTK version 9.2.6 instead of 9.3 to see if that fixes the issue on Windows 10?