CadQuery/ocp-build-system #30: CI VTK Builds Failing on Windows

jmwright, 2024-06-14:
Thanks for the help getting curl to work @jdegenstein 

The VTK builds on Windows are still broken because of missing tools. The reason that I am pursuing this is that the official VTK 9.2.x wheels only go up to Python 3.11.  We're going to have to support our own VTK wheels until the time when OCP catches up to the VTK 9.3.x line, and even then VTK may lag behind the latest version of Python.

I don't want to re-bundle VTK with OCP like we did before because it causes other problems.

@jdegenstein If you have any ideas on how to get the Windows build of VTK working, please feel free. You have been added as a collaborator.

jdegenstein, 2024-06-17:
I was finally able to get Windows builds working here: https://github.com/jdegenstein/ocp-build-system/actions/runs/9532941782

I was also able to get MacOS + arm64 builds working. Now I just need to combine the different approaches with minimum code duplication and bring to this (the main) repo.

jmwright, 2024-06-17:
Thanks for the efforts @jdegenstein !

jdegenstein, 2024-06-17:
Alright, I brought over my changes to the VTK build process. There were a lot of changes, and a lot of these were platform-specific unfortunately.

The succeeding runs -- https://github.com/CadQuery/ocp-build-system/actions/runs/9550829968

The above includes wheels for Python 3.9 through 3.12 against Linux, MacOS-13 (x86_64), MacOS-14, and Windows.

I had some trouble with MacOS-11 runners, so I updated to MacOS-13.

More information available at the merged PR here https://github.com/CadQuery/ocp-build-system/pull/31

jmwright, 2024-06-17:
Thanks @jdegenstein ! Starting to test on Debian now.

jmwright, 2024-06-17:
@jdegenstein VTK wheel works fine on Debian Bookworm on AMD64 and I get a not supported error on MacOS 10.14.6 (Intel-based version).

jdegenstein, 2024-06-18:
This page indicates that macOS 10.13 is the minimum allowed target currently for Xcode 14.1 through 15.3 (the latest). https://developer.apple.com/support/xcode/

The macos-13 runners include Xcode 15.2 through 14.1 currently (and a few versions in between).
https://github.com/actions/runner-images/blob/macos-13/20240609.1/images/macos/macos-13-Readme.md

jmwright, 2024-06-18:
@jdegenstein I did not realize that the Mac I had access to was so out of date. I was thinking it that the naming was based on the middle number, but I see now that is incorrect. I would say let's not worry about such old versions.

jdegenstein, 2024-06-18:
@jmwright I think you raise a valuable point, I don't like to abandon older platforms prematurely. I went ahead and tested `export MACOSX_DEPLOYMENT_TARGET=10.13`.

The resulting wheels are here, I would try the macos-13 build first. https://github.com/jdegenstein/ocp-build-system/actions/runs/9566934154

If this doesn't work then probably not worth spending a lot of time on it. There are some warnings during build that make me think it won't work on anything older than macos 11.0.

jmwright, 2024-06-18:
@jdegenstein `pip` does not complain about the platform being unsupported during the install, but there is a segfault when I try either `import vtk` or `from vtk import *`.  I agree that it is not worth spending a lot of time on this since the platform is so old.

I have not had an overwhelming response in the past when I have asked the community to test wheels like this. I can create a cadquery-vtk project on PyPI and upload these wheels, but it will be hard to verify that they work until people start installing the new cadquery-ocp package that relies on these. Not an ideal position to be in, but I am not sure we have a lot of choice.

OCP 7.7.2.1 has been released I believe, so we can use it as a test case before releasing cadquery-ocp 7.8.0 and (eventually) 7.8.1.

Thoughts?

jdegenstein, 2024-06-19:
@jmwright I agree it is time to move on from MacOS < 11.0. Regarding the other wheels, I think we should lean on testing them in CI a bit. At a minimum we should `import vtk` but we could do more than that, if appropriate. I don't have much knowledge about VTK so I am not sure what would be best here.

jdegenstein, 2024-06-22:
@jmwright ok I did a simple import test here https://github.com/jdegenstein/ocp-build-system/actions/runs/9619686072/job/26536337602 and it worked (only done on py311 and ubuntu-20.04) but it shouldn't be too hard to do it here as well

jmwright, 2024-06-27:
@jdegenstein That sounds good.

closed 2024-09-06 by jmwright