CadQuery/ocp-build-system #33: Strip OCP.so for smaller installed footprint

fischman, 2024-07-24:
`pip install build123d` leads to an `OCP.cpython-310-x86_64-linux-gnu.so` sized 242MiB.
Running `strip` on that `.so` reduces it to 143MiB, and all the cadquery and build123d tests still pass. See details on [discord](https://discord.com/channels/964330484911972403/1074840524181217452/1264637057452605593), as well as a motivation for this (service provider imposed limits on image size.

Could the OCP .so be stripped before inclusion in the wheel? (possibly [here](https://github.com/CadQuery/ocp-build-system/blob/4ca7187075f62253f5dcedf58890bd054f6eb728/setup.py#L117) though I've just skimmed the repo)

jmwright, 2024-08-13:
@fischman Is there log output from when you ran the `strip` command which shows what is being done to the .so file? I'm hesitant to do this in CI without a better understanding of what it is doing to the lib file.

fischman, 2024-08-13:
@jmwright

> Is there log output from when you ran the strip command which shows what is being done to the .so file?

No; `strip` is silent by default. When given `-v` it emits something like `copy from `/tmp/OCP.cpython-310-x86_64-linux-gnu.so' [elf64-x86-64] to `/tmp/stGEw0qJ' [elf64-x86-64]` but IDK that that's useful.

> I'm hesitant to do this in CI

Possibly debian's guidance will increase confidence https://www.debian.org/doc/debian-policy/ch-files.html#shared-libraries:~:text=All%20installed%20shared%20libraries%20should%20be%20stripped%20with

> a better understanding of what it is doing to the lib file.

In terms of ELF sections, this shows that the `.comment`, `.strtab`, and `.symtab` sections are removed:
```
$ for suffix in before after; do l="OCP.cpython-310-x86_64-linux-gnu.so.$suffix"; ls -l $l; readelf --sections $l | sed -ne 's/^ *\[[ 0-9]*\] *\([^ ]*\) .*/\1/p'|sort > sections.$suffix; done; diff sections.before sections.after
.rwxrwxr-x 249M ami 13 Aug 15:36 OCP.cpython-310-x86_64-linux-gnu.so.before
.rwxrwxr-x 144M ami 13 Aug 15:37 OCP.cpython-310-x86_64-linux-gnu.so.after
2d1
< .comment
26,27d24
< .strtab
< .symtab
```
(the above is using debian's recommended flags: `strip --strip-unneeded --remove-section=.comment --remove-section=.note`)

`readelf --sections` on the "before" lib shows the sections account for: `.comments` is 84 bytes, `.symtab` is 11,309,856 bytes, and `.symtab` is 90,439,979 bytes.

Happy to answer any other questions I can.

jmwright, 2024-09-10:
We can discuss targeting 7.8.1 for this. A PR to add this to CI and the local build scripts would be welcome.

fischman, 2024-09-10:
@jmwright happy to send a PR but won't be at a computer for another week.

fischman, 2024-09-17:
@jmwright https://github.com/CadQuery/ocp-build-system/pull/38 has a first crack at this.
I was mostly guessing as to how this repo works :)

closed 2024-10-08 by jmwright