CadQuery/ocp-build-system #34: Release OCP 7.7.2.1

akoen, 2024-07-30:
Hi!

I see that OCP version 7.7.2.1 was released in May. I'd really like to use it's functionality (IMesh_Tools) in my meshing package https://github.com/stellarmesh/stellarmesh. Is there a plan or timeline for releasing this version on PyPI?

Thanks!

jmwright, 2024-07-30:
The timeline is as soon as possible. We've been fighting some build issues though. Hopefully within the next week or so we will be able to release 7.7.2.1.

akoen, 2024-07-30:
Fantastic @jmwright great to hear

jmwright, 2024-09-06:
All - Please download the correct wheel for your platform and Python version and do a test installation. Once these have been tested, I will publish them to PyPI. During the installation, cadquery-vtk will be downloaded from PyPI since it is no longer embedded inside the OCP wheel.

https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.1

jmwright, 2024-09-10:
This has been completed.

https://pypi.org/project/cadquery-ocp/

jmwright, 2024-09-11:
There was an issue with the VTK dependency that is causing this wheel not to work properly for certain use cases, so I am reopening this issue.

akoen, 2025-01-29:
Works great on my end!

jmwright, 2025-01-29:
@akoen Thanks for reporting back.

closed 2025-01-29 by jmwright