CadQuery/ocp-build-system #41: cadquery_ocp Python Wheel requires glibc 2.35+

alexanderilyin, 2024-12-13:
`poetry install ...` results in this on Debian 11 Bullseye:

```
Skipping wheel cadquery_ocp-7.7.2-cp310-cp310-manylinux_2_35_x86_64.whl as this is not supported by the current environment
```

Since Bullseye long-term support ends 31 August 2026 I wonder if it's possible to support earlier version of glibc which is shipped with Debian 11?

jmwright, 2024-12-23:
@alexanderilyin Can you test this beta release out?
```
pip install --pre cadquery_ocp==7.7.2.2b2
```

alexanderilyin, 2025-01-09:
@jmwright unfortunately installation still fails:

- [install.txt](https://github.com/user-attachments/files/18359447/install.txt)

```
$ ldd --version
ldd (Debian GLIBC 2.31-13+deb11u11) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

```
$ lsb_release -d
Description:    Debian GNU/Linux 11 (bullseye)
```

alexanderilyin, 2025-01-09:
I switch to Bookworm and it works just fine:

```
$ ldd --version
ldd (Debian GLIBC 2.36-9+deb12u9) 2.36
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

```
$ lsb_release -d
No LSB modules are available.
Description:    Debian GNU/Linux 12 (bookworm)
```

```
$ pip install cadquery-ocp>=7.7.0
$ pip freeze | grep cadquery
cadquery-ocp==7.7.2
```

jdegenstein, 2025-01-29:
@alexanderilyin FYI cadquery-ocp 7.8+ was recently released and is marked with manylinux_2_31 which I believe is Debian 11 compatible. Can you confirm and close if this solves your issue?

https://pypi.org/project/cadquery-ocp/7.8.1.1.post1/#files